KinDR007/VictronMPPT-ESPHOME #183: Smart Solar MPPT 150/70

meccip, 2024-11-11:
Hello. I am using your code with smartsolar 150/70 and i have some difficulties. I am not receiving all sensors on every 3seconds. Only at first start all sensors are present, then i only receive 6 (battery v, charging mode, pv voltage, pv current, pv power, tracking mode) After a period of time, sensors switch to other 7 (day number,max power yesterday,today, error and yield today,yesterday,total)
I try with esp01 and nodemcu, i changed throttle interval from none to 10s, i changed rx pins. I still have the same result.

Thank you.

syssi, 2024-11-11:
Could you enable the debug output of the UART component and provide some logs?

https://github.com/KinDR007/VictronMPPT-ESPHOME/blob/main/debug-esp8266-example.yaml#L31

meccip, 2024-11-11:
Hey, syssi. I'm glad to hear from you.
I have a file with log
[uart_log.txt](https://github.com/user-attachments/files/17700509/uart_log.txt)

syssi, 2024-11-11:
These are the important parts from the log:

```
[11:46:34][D][uart_debug:158]: <<< ":A0202000200000045\n"
[11:46:34][D][uart_debug:158]: <<< "\r\n"
[11:46:34][D][uart_debug:158]: <<< "ERR\t0\r\n"
[11:46:34][D][uart_debug:158]: <<< "LOAD\tON\r\n"
[11:46:34][D][uart_debug:158]: <<< "Relay\tOFF\r\n"
[11:46:34][D][uart_debug:158]: <<< "H19\t1158\r\n"
[11:46:34][D][uart_debug:158]: <<< "H20\t77\r\n"
[11:46:34][D][uart_debug:158]: <<< "H21\t479\r\n"
[11:46:34][D][uart_debug:158]: <<< "H22\t135\r\n"
[11:46:34][D][uart_debug:158]: <<< "H23\t525\r\n"
[11:46:34][D][uart_debug:158]: <<< "HSDS\t2\r\n"
[11:46:34][D][uart_debug:158]: <<< "Checksum\t\x05:AD5ED009614DF\n"

[11:46:34][D][uart_debug:158]: <<< ":A8DED00961427\n"
[11:46:34][D][uart_debug:158]: <<< ":A0002000148\n"
[11:46:35][D][uart_debug:158]: <<< "\r\n"

[11:46:35][D][uart_debug:158]: <<< "PID\t0xA06C\r\n"
[11:46:35][D][uart_debug:158]: <<< "FWE\t316FF\r\n"
[11:46:35][D][uart_debug:158]: <<< "SER#\tHQ2247EP3FN\r\n"
[11:46:35][D][uart_debug:158]: <<< "V\t52700\r\n"
[11:46:35][D][uart_debug:158]: <<< "I\t7600\r\n"
[11:46:35][D][uart_debug:158]: <<< "VPV\t103870\r\n"
[11:46:35][D][uart_debug:158]: <<< "PPV\t412\r\n"
[11:46:35][D][uart_debug:158]: <<< "MPPT\t2\r\n"
[11:46:35][D][uart_debug:158]: <<< "CS\t3\r\n"
[11:46:35][D][uart_debug:158]: <<< "OR\t0x00000000\r\n"

[11:46:35][D][uart_debug:158]: <<< "Checksum\t\xFA"
[11:46:35][D][uart_debug:158]: <<< ":A2001002470030093\n"
[11:46:35][D][uart_debug:158]: <<< ":ADBED00BF0ABA\n"
[11:46:35][D][uart_debug:158]: <<< ":ABCED0045A10000BC\n"
[11:46:35][D][uart_debug:158]: <<< ":ABBED008D28EE\n"
[11:46:35][D][uart_debug:158]: <<< ":AB4ED000DA10000FC\n"
[11:46:35][D][uart_debug:158]: <<< ":AB5ED00802801\n"
[11:46:35][D][uart_debug:158]: <<< "\r\n"
[11:46:35][D][uart_debug:158]: <<< "ERR\t0\r\n"
[11:46:35][D][uart_debug:158]: <<< "LOAD\tON\r\n"
[11:46:35][D][uart_debug:158]: <<< "Relay\tOFF\r\n"
[11:46:35][D][uart_debug:158]: <<< "H19\t1158\r\n"
[11:46:35][D][uart_debug:158]: <<< "H20\t77\r\n"
[11:46:35][D][uart_debug:158]: <<< "H21\t479\r\n"
[11:46:35][D][uart_debug:158]: <<< "H22\t135\r\n"
[11:46:35][D][uart_debug:158]: <<< "H23\t525\r\n"
[11:46:35][D][uart_debug:158]: <<< "HSDS\t2\r\n"
[11:46:35][D][uart_debug:158]: <<< "Checksum\t\x05"
[11:46:36][D][uart_debug:158]: <<< "\r\n"

[11:46:36][D][uart_debug:158]: <<< "PID\t0xA06C\r\n"
[11:46:36][D][uart_debug:158]: <<< "FWE\t316FF\r\n"
[11:46:36][D][uart_debug:158]: <<< "SER#\tHQ2247EP3FN\r\n"
[11:46:36][D][uart_debug:158]: <<< "V\t52700\r\n"
[11:46:36][D][uart_debug:158]: <<< "I\t7600\r\n"
[11:46:36][D][uart_debug:158]: <<< "VPV\t103880\r\n"
[11:46:36][D][uart_debug:158]: <<< "PPV\t412\r\n"
[11:46:36][D][uart_debug:158]: <<< "MPPT\t2\r\n"
[11:46:36][D][uart_debug:158]: <<< "CS\t3\r\n"
[11:46:36][D][uart_debug:158]: <<< "OR\t0x00000000\r\n"
[11:46:36][D][uart_debug:158]: <<< "Checksum\t\xF9:A2001002570030092\n"

[11:46:36][D][uart_debug:158]: <<< ":ADBED00D10AA8\n"
[11:46:36][D][uart_debug:158]: <<< ":ABCED0024A10000DD\n"
[11:46:36][D][uart_debug:158]: <<< ":ABBED009C28DF\n"
[11:46:36][D][uart_debug:158]: <<< ":AB4ED00BEA100004B\n"
[11:46:36][D][uart_debug:158]: <<< ":AB5ED00802801\n"
[11:46:36][D][uart_debug:158]: <<< ":AD5ED009714DE\n"
[11:46:36][D][uart_debug:158]: <<< ":A8DED00971426\n"
[... 150 lines cut ...]
[11:46:41][D][uart_debug:158]: <<< "CS\t3\r\n"
[11:46:41][D][uart_debug:158]: <<< "OR\t0x00000000\r\n"
[11:46:42][D][uart_debug:158]: <<< "Checksum\t\xFC"
[11:46:42][D][uart_debug:158]: <<< "\r\n"
[11:46:42][D][uart_debug:158]: <<< "ERR\t0\r\n"
[11:46:42][D][uart_debug:158]: <<< "LOAD\tON\r\n"
[11:46:42][D][uart_debug:158]: <<< "Relay\tOFF\r\n"
[11:46:42][D][uart_debug:158]: <<< "H19\t1158\r\n"
[11:46:42][D][uart_debug:158]: <<< "H20\t77\r\n"
[11:46:42][D][uart_debug:158]: <<< "H21\t479\r\n"
[11:46:42][D][uart_debug:158]: <<< "H22\t135\r\n"
[11:46:42][D][uart_debug:158]: <<< "H23\t525\r\n"
[11:46:42][D][uart_debug:158]: <<< "HSDS\t2\r\n"
[11:46:42][D][uart_debug:158]: <<< "Checksum\t\x05:A8EED0093013C\n"
[11:46:42][D][uart_debug:158]: <<< ":A2001002B7003008C\n"
[11:46:42][D][uart_debug:158]: <<< ":ADBED00C80AB1\n"
[11:46:42][D][uart_debug:158]: <<< ":ABCED00DEA0000024\n"
[11:46:42][D][uart_debug:158]: <<< ":ABBED00AA28D1\n"
[11:46:42][D][uart_debug:158]: <<< ":AB4ED00B1A0000059\n"
[11:46:42][D][uart_debug:158]: <<< ":AB5ED008228FF\n"
[11:46:42][D][uart_debug:158]: <<< ":A8EED0092013D\n"
[11:46:42][D][uart_debug:158]: <<< "\r\n"
```

I have to think about it. The behavior of your victron device is a bit strange. Did you try to update the firmware already?

meccip, 2024-11-11:
I just bought it. When i connect to it, force me to update.
I try to get data with nodered and ve.direct. I made a FTDI cable and test it with my phone (i can connect to controller via app, receive data and make some settings, no bluetooth active only wire). (after that i realize that i could buy a bluesolar without bluetooth and use the phone with cable to set the controller )

I use ve.direct nodered but i have no connection, then i try serial flow and i get some buffer data every second. I am still working on it.

Anyway, thank you for your help.

meccip, 2024-11-11:
i just want to add that i look into app controller product info and is firmware Rev3

syssi, 2024-11-13:
Okay. I will try to reproduce the issue locally by replaying your captured data to get an better idea what's going on. This will take some time.

meccip, 2024-11-15:
if i set throttle at 6seconds (lower than that issue is present) i have the data but only 6 sensors received at 6 seconds. So, could be too much data to process if is working at 3 seconds.
It is possible to process at 3sec only pv voltage,current and power and the rest after 10sec ?

syssi, 2024-11-15:
Please avoid the `throttle` option because it expects one big messages containing all measurements. Your device sends two messages periodically. The second gets always discarded by the throttle.

meccip, 2024-11-15:
i remove throttle and i only receive 7 sensors ( day_number, error and history sensors )

syssi, 2024-11-15:
Could you make sure your `victron` section looks like this and provide another log?

```
victron:
  uart_id: uart_0
  id: victron0
  throttle: 0s
```

If I emulate your victron device using this YAML on a second ESP: https://github.com/KinDR007/VictronMPPT-ESPHOME/pull/186/files

I'm able to decode these measurements periodically (`throttle: 0s` required):

```
[09:26:11][D][sensor:093]: 'victron-mppt battery voltage': Sending state 52.70000 V with 3 decimals of accuracy
[09:26:11][D][sensor:093]: 'victron-mppt panel current': Sending state 7.60000 A with 3 decimals of accuracy
[09:26:11][D][sensor:093]: 'victron-mppt panel voltage': Sending state 103.87000 V with 3 decimals of accuracy
[09:26:11][D][sensor:093]: 'victron-mppt panel power': Sending state 412.00000 W with 0 decimals of accuracy
[09:26:11][D][sensor:093]: 'victron-mppt tracking mode id': Sending state 2.00000  with 0 decimals of accuracy
[09:26:11][D][text_sensor:064]: 'victron-mppt tracking mode': Sending state 'Active'
[09:26:11][D][sensor:093]: 'victron-mppt charging mode id': Sending state 3.00000  with 0 decimals of accuracy
[09:26:11][D][text_sensor:064]: 'victron-mppt charging mode': Sending state 'Bulk'
[09:26:11][D][sensor:093]: 'victron-mppt error code': Sending state 0.00000  with 0 decimals of accuracy
[09:26:11][D][text_sensor:064]: 'victron-mppt error': Sending state 'No error'
[09:26:11][D][sensor:093]: 'victron-mppt yield total': Sending state 11580.00000 Wh with 0 decimals of accuracy
[09:26:11][D][sensor:093]: 'victron-mppt yield today': Sending state 770.00000 Wh with 0 decimals of accuracy
[09:26:11][D][sensor:093]: 'victron-mppt max power today': Sending state 479.00000 W with 0 decimals of accuracy
[09:26:11][D][sensor:093]: 'victron-mppt yield yesterday': Sending state 1350.00000 Wh with 0 decimals of accuracy
[09:26:11][D][sensor:093]: 'victron-mppt max power yesterday': Sending state 525.00000 W with 0 decimals of accuracy
[09:26:11][D][sensor:093]: 'victron-mppt day number': Sending state 2.00000  with 0 decimals of accuracy
```

How does it look like on your site?

syssi, 2024-11-15:
```
      - uart.write: "PID\t0xA06C\r\n"        # Device typ gets published once
      - uart.write: "FWE\t316FF\r\n"         # Firmware version gets published once
      - uart.write: "SER#\tHQ2247EP3FN\r\n"  # Serial number gets published once
      - uart.write: "V\t52700\r\n"           # battery voltage
      - uart.write: "I\t7600\r\n"            # panel current
      - uart.write: "VPV\t103870\r\n"        # panel voltage
      - uart.write: "PPV\t412\r\n"           # panel power
      - uart.write: "MPPT\t2\r\n"            # tracking mode
      - uart.write: "CS\t3\r\n"              # charging mode
      - uart.write: "OR\t0x00000000\r\n"     # off_reason (missing at the YAML?)
      - uart.write: "Checksum\t\xFA"         # ignored
      - uart.write: ":A2001002470030093\n"   # unsupported/ignored
      - uart.write: ":ADBED00BF0ABA\n"       # unsupported/ignored
      - uart.write: ":ABCED0045A10000BC\n"   # unsupported/ignored
      - uart.write: ":ABBED008D28EE\n"       # unsupported/ignored
      - uart.write: ":AB4ED000DA10000FC\n"   # unsupported/ignored
      - uart.write: ":AB5ED00802801\n"       # unsupported/ignored
      - uart.write: "\r\n"
      - uart.write: "ERR\t0\r\n"             # error
      - uart.write: "LOAD\tON\r\n"           # load_state
      - uart.write: "Relay\tOFF\r\n"         # relay_state
      - uart.write: "H19\t1158\r\n"          # yield_total
      - uart.write: "H20\t77\r\n"            # yield_today
      - uart.write: "H21\t479\r\n"           # max_power_today
      - uart.write: "H22\t135\r\n"           # yield_yesterday
      - uart.write: "H23\t525\r\n"           # max_power_yesterday
      - uart.write: "HSDS\t2\r\n"            # day_number
      - uart.write: "Checksum\t\x05"         # ignored
      - uart.write: "\r\n"
```

meccip, 2024-11-15:
it is working now. i set throttle to 0s. i make a log.

[uart_log.txt](https://github.com/user-attachments/files/17771926/uart_log.txt)

syssi, 2024-11-15:
The `throttle` feature/implementation uses the `Checksum` line as "complete frame indicator". Your firmware has two checksums per frame which breaks the behaviour.

syssi, 2024-11-15:
This is a duplicate of #96.

closed 2024-11-15 by syssi

meccip, 2024-12-05:
The code is running good (i add it some average readings because data was going crazy). Tested with esp01, d1 mini and nodemcu

But now i have a new issue. Because esp is near solar controller, i was thinking to add in the code 4 temperature sensors to monitor pv and battery fuses and victron heatsink  and a relay to shutdown controller.
I use ADS1115 voltage reader to convert NTC to temperature that work with esp on I2C.
So, i added to code: ads1115, resistance, ntc and switch platforms and initiate I2C bus.

And now, after 4 minutes esp goes offline, mqtt is not sending data but esp still running the code (because i made led to blink when data is received and processed - and is blinking). After 3-4 minutes mqtt start again.
I use d1mini with 4M of ram. At upload i have 56% use of ram. I suspect that wifi is going down because is to much code in the memory. (i have the same situation with an esp01 1M of ram, when i use it with jk bms to get too many sensors and control switch).

I think is too much data for my wemos. I read that D1 mini PRO have 16M of ram. Do you think this will solve my problem ?

KinDR007, 2024-12-05:
Hi,
Move to esp32.
Esp8266 has 80kb of ram
Esp32 has 520kb of ram 

16Mb it is not RAM it is flash !

meccip, 2024-12-05:
get it now. so on esp32 i can run my modified code ?

KinDR007, 2024-12-06:
> get it now. so on esp32 i can run my modified code ?

Simply? Yes ,change gpio pin's name from esp8266 to esp32 change board name on top of yaml and upload to esp32 😉

meccip, 2024-12-06:
ok. i order [wemos S2 mini](https://www.wemos.cc/en/latest/s2/s2_mini.html) because it have to fit into my project to replace D1 mini. I hope is esp32. Until then i manage to make code work on d1 mini. I put into mqtt platform a reset function on disconnect, so when mqtt stop sending info, will reset board.