KinDR007/VictronMPPT-ESPHOME #187: victron-mppt panel current output is not panel current

strupp666, 2024-11-26:
Hello,
in smartsolar-mppt-esp8266-example.yaml I get the the wrong value of victron-mppt panel current.
In fact it is battery current (measured by amperemeter).
Both values would be interesting for me.
Thank you in advance.

strupp666, 2024-11-26:
Okay, I do not know which measurements are available with vedirect. Via Bluetooth App you can see Panel-Power, Panel-Voltage, Panel-Current - Battery-Current, Battery-Voltage.
In HA I can see victron-mppt load current, which is unknown. I think this is the correct name for this value.
Just as a small note. Thank you very much for the project and the work.

strupp666, 2024-11-26:
FYI: Blue Solar MPPT 150/35 charger

syssi, 2024-11-26:
If you enable the `debug` mode of the `uart` component:

```
  debug:
    direction: BOTH
    dummy_receiver: false
    after:
      delimiter: "\n"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);
```

You will see the complete traffic at the log. Could you post a log with debugging enabled?

syssi, 2024-11-26:
See https://github.com/KinDR007/VictronMPPT-ESPHOME/issues/146

strupp666, 2024-11-26:
[logs_victron-esp8266_logs.txt](https://github.com/user-attachments/files/17920307/logs_victron-esp8266_logs.txt)

strupp666, 2024-11-26:
raspberry 3b ist too small

syssi, 2024-11-26:
This is a complet RAW frame (chunk of measurements) of your device:

```
<<< "PID\t0xA058\r\n"
<<< "FW\t164\r\n"
<<< "SER#\tHQ2311G69FY\r\n"
<<< "V\t52920\r\n"
<<< "I\t3800\r\n"
<<< "VPV\t84750\r\n"
<<< "PPV\t206\r\n"
<<< "CS\t3\r\n"
<<< "MPPT\t2\r\n"
<<< "OR\t0x00000000\r\n"
<<< "ERR\t0\r\n"
<<< "LOAD\tON\r\n"
<<< "H19\t3675\r\n"
<<< "H20\t239\r\n"
<<< "H21\t725\r\n"
<<< "H22\t88\r\n"
<<< "H23\t727\r\n"
<<< "HSDS\t25\r\n"
<<< "Checksum\t\x1D"
```

If you replace `\t` with a tab and `\r\n` with a line break it looks like this:

```
PID     0xA058
FW      164
SER#    Q2311G69FY
V       52920
I       3800
VPV     84750
PPV     206
CS      3
MPPT    2
OR      0x00000000
ERR     0
LOAD    ON
H19     3675
H20     239
H21     725
H22     88
H23     727
HSDS    25
Checksum XXX
```

Please lookup the meaning of the keys at the documentation here: https://github.com/KinDR007/VictronMPPT-ESPHOME?tab=readme-ov-file#sensors

Missing values/keys cannot be passed to your Home Automation because the measurement isn't available / part of the frame.

strupp666, 2024-11-26:
Okay I understand. But I don´t understand the wrong naming in HA. 
[image]
 
49/79,85=0,613 
0,9 A ist Battery current.

strupp666, 2024-11-26:
Thank you. I will calculate panel current in nodeRed.

closed 2024-11-26 by strupp666

strupp666, 2024-11-26:
Yes, I know, but as a beginner it is easier for me to visualize it with nodeRed.