KinDR007/VictronMPPT-ESPHOME #188: Question

RFDarter, 2024-11-29:
Is there a way to also control the mppt,?
I have a remote setup that needs every Wh it can get, so installing a raspi with venus os would consume to much power.

It consists of the battery, mppt, lte-router, wlan camera.
So I could simply add a esp which sends data to my home and do all the calculations and control there.

syssi, 2024-11-29:
This implementation is read-only. The victron device transmits a ASCII message periodically. This project is just a decoder. The HEX protocol needs to be implemented forbread-write operations.

closed 2024-12-05 by KinDR007

RFDarter, 2024-12-05:
> This implementation is read-only. The victron device transmits a ASCII message periodically. This project is just a decoder. The HEX protocol needs to be implemented forbread-write operations.

I guess nobody has done that already?

KinDR007, 2024-12-06:
This feature it's not planned