KinDR007/VictronMPPT-ESPHOME #197: Last transmission too long ago

gw942400, 2025-03-17:
I have 2 victron smart solar mppt

1.100/30 : the script run well
2. 150/35 : i have this message on the log 

[15:26:28][W][victron:093]: Last transmission too long ago

i check the wiring , it is the same configuration 

do you have an idea ?

thank you for your help

KinDR007, 2025-03-17:
Hi, it looks just like this one.

https://github.com/KinDR007/VictronMPPT-ESPHOME/issues/152

Guy559, 2026-02-24:
Happening for me as well. Plenty of good data but the "Last transmission too long ago" thrown out every few seconds. Logs below. Is this a data dropout from the ve.direct port? Wiring is fine. Have a splitter and the other device reading from ve.direct port (off a SmartShunt) has no issues with data. Thanks!

[21:39:13.512][D][text_sensor:113]: 'victron-smartshunt alarm reason' >> 'No error'
[21:39:13.512][D][text_sensor:113]: 'victron-smartshunt model description' >> 'SmartShunt 500A/50mV'
[21:39:13.529][D][sensor:118]: 'victron-smartshunt dc monitor mode id' >> 0 
[21:39:13.531][D][text_sensor:113]: 'victron-smartshunt dc monitor mode' >> 'Battery monitor (BMV)'
[21:39:14.281][W][victron:093]: Last transmission too long ago
[21:39:15.729][W][victron:093]: Last transmission too long ago
[21:39:16.325][W][victron:093]: Last transmission too long ago
[21:39:18.934][D][sensor:118]: 'victron-smartshunt depth of the deepest discharge' >> -1241.226 Ah
[21:39:18.951][D][sensor:118]: 'victron-smartshunt depth of the last discharge' >> -751.285 Ah
[21:39:18.953][D][sensor:118]: 'victron-smartshunt depth of the average discharge' >> -696.104 Ah
[21:39:18.953][D][sensor:118]: 'victron-smartshunt number of charge cycles' >> 36 
[21:39:18.964][D][sensor:118]: 'victron-smartshunt number of full discharges' >> 0 
[21:39:18.971][D][sensor:118]: 'victron-smartshunt cumulative amp hours drawn' >> -26743.090 Ah
[21:39:18.974][D][sensor:118]: 'victron-smartshunt min battery voltage' >> 15.148 V
[21:39:18.978][D][sensor:118]: 'victron-smartshunt max battery voltage' >> 58.152 V
[21:39:18.993][D][sensor:118]: 'victron-smartshunt last full charge' >> 4424 min
[21:39:18.993][D][sensor:118]: 'victron-smartshunt number of automatic synchronizations' >> 3 
[21:39:18.996][D][sensor:118]: 'victron-smartshunt number of low main voltage alarms' >> 0 
[21:39:18.996][D][sensor:118]: 'victron-smartshunt number of high main voltage alarms' >> 0 
[21:39:18.996][D][sensor:118]: 'victron-smartshunt min auxiliary battery voltage' >> 0.000 V
[21:39:19.064][D][sensor:118]: 'victron-smartshunt max auxiliary battery voltage' >> 0.000 V
[21:39:19.064][D][sensor:118]: 'victron-smartshunt amount of discharged energy' >> 1407210 Wh
[21:39:19.064][D][sensor:118]: 'victron-smartshunt amount of charged energy' >> 1448750 Wh
[21:39:20.249][W][victron:093]: Last transmission too long ago
[21:39:24.412][D][victron:1105]: Unhandled property:  

[21:39:24.431]V 52688
[21:39:24.434][D][sensor:118]: 'victron-smartshunt battery current' >> -55.639 A
[21:39:24.434][D][sensor:118]: 'victron-smartshunt instantaneous power' >> -2932 W
[21:39:24.434][D][sensor:118]: 'victron-smartshunt consumed amp hours' >> -283.320 Ah
[21:39:24.504][D][sensor:118]: 'victron-smartshunt state of charge' >> 78 %
[21:39:24.504][D][sensor:118]: 'victron-smartshunt time to go' >> 966 min
[21:39:24.504][D][text_sensor:113]: 'victron-smartshunt alarm condition active' >> 'OFF'
[21:39:24.510][D][text_sensor:113]: 'victron-smartshunt alarm reason' >> 'No error'
[21:39:24.510][D][text_sensor:113]: 'victron-smartshunt model description' >> 'SmartShunt 500A/50mV'
[21:39:24.706][W][victron:093]: Last transmission too long ago

[21:39:24.934]H1 -1241226
[21:39:24.935][D][sensor:118]: 'victron-smartshunt depth of the last discharge' >> -751.285 Ah
[21:39:24.935][D][sensor:118]: 'victron-smartshunt depth of the average discharge' >> -696.104 Ah

Guy559, 2026-02-25:
Things are significantly improved. Read through some other issue suggestions. Dropped threshold to 0s which didn’t make much of a difference. What DID seem to make a different was commenting out most of the reported statistics. Realistically I only need ~7-9 of the items, so I commented out anything not required. As you can see in the chart below, the dropouts decreased substantially during past 12 hours or so.

[image]