KinDR007/VictronMPPT-ESPHOME #201: text_sensor is deprecated and will be removed ...

Binifada1956, 2025-05-24:
Hello, this message appeared on the monitor today:

_INFO ESPHome 2025.5.0
INFO Reading configuration /config/esphome/esphome-web-ec6730.yaml...
INFO Updating https://github.com/KinDR007/VictronMPPT-ESPHOME.git@main
WARNING Using `text_sensor.TEXT_SENSOR_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `text_sensor.text_sensor_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: text_sensor_

Roving-Ronin, 2025-05-24:
Suggest checking existing issues before opening new ones, the top of the existing issues is for this same issue.

https://github.com/KinDR007/VictronMPPT-ESPHOME/issues/199

Please close this duplicate.

syssi, 2025-05-24:
Duplicate of #199

closed 2025-05-24 by syssi