KinDR007/VictronMPPT-ESPHOME #210: WARNING Using text_sensor.TEXT_SENSOR_SCHEMA

Michl3088, 2025-08-01:
ESPHome shows me the following error:
WARNING Using `text_sensor.TEXT_SENSOR_SCHEMA` is deprecated and will be removed in ESPHome 2025.11.0. Please use `text_sensor.text_sensor_schema(...)` instead. If you are seeing this, report an issue to the external_component author and ask them to update it. https://developers.esphome.io/blog/2025/05/14/_schema-deprecations/. Component using this schema: text_sensor

Will the sensor be updated?

syssi, 2025-08-01:
It's already fixed by https://github.com/KinDR007/VictronMPPT-ESPHOME/commit/0e5efa1497a2a6ef4db8348fe82e1507f4ba3e19

Do you use an old copy of the external component or an outdated version?

syssi, 2025-08-01:
You are using two external components:

```
  external_components_source1: github://KinDR007/VictronMPPT-ESPHOME@main
  external_components_source2: github://KinDR007/VictronSmartShunt-ESPHOME@main
```

The second one wasn't updated yet.

KinDR007, 2025-08-01:
> ```
> substitutions:
>   external_components_source1: github://KinDR007/VictronMPPT-ESPHOME@main
>   external_components_source2: github://KinDR007/VictronSmartShunt-ESPHOME@main
>   lower_devicename_mppt: "smartsolar"
>   lower_devicename_shunt: "smartshunt"
>   devicename_mppt: "smartsolar-150-35"
>   devicename_shunt: "smartshunt-300A"
>   config_version_mppt: "v2024.06.21"
>   config_version_shunt: "v2021.06.02.001"
>   wifi_fast_connect: "false"
>   accuracy: "2"
> 
> esphome:
>   name: "${lower_devicename_mppt}-${lower_devicename_shunt}"
>   #platform: ESP32
>   #board: nodemcu-32s
>   min_version: 2024.6.0
> 
> esp32:
>   board: nodemcu-32s
>   framework: 
>     type: esp-idf  
> 
> external_components:
>   - source: ${external_components_source1}
>     refresh: 0s
>   - source: ${external_components_source2}
>     refresh: 0s
> 
> wifi:
>   ssid: !secret wifi_ssid
>   password: !secret wifi_password
>   power_save_mode: none
>   fast_connect: $wifi_fast_connect
>   manual_ip:
>     static_ip: 192.168.178.133
>     gateway: 192.168.178.1
>     subnet: 255.255.255.0
>   ap:
>     ssid: "Fallback victron"
>     password: !secret wifi_password
> 
> captive_portal:
> 
> ota:
>   platform: esphome
>   password: "XXX"
> 
> logger:
>   baud_rate: 0
> 
> api:
>   encryption:
>     key: "XXX"
> 
> web_server:
>   port: 80
> 
> time:
>   - platform: sntp
> 
> # UART-Konfiguration für beide Geräte
> uart:
>   - id: uart_bus_mppt
>     tx_pin: GPIO16
>     rx_pin: GPIO17
>     baud_rate: 19200
>     stop_bits: 1
>     data_bits: 8
[... 225 lines cut ...]
>       int hrs = 0;
>       int mnts = 0;
>       if (dur > 86399) {
>         dys = trunc(dur / 86400);
>         dur = dur - (dys * 86400);
>       }
>       if (dur > 3599) {
>         hrs = trunc(dur / 3600);
>         dur = dur - (hrs * 3600);
>       }
>       if (dur > 59) {
>         mnts = trunc(dur / 60);
>         dur = dur - (mnts * 60);
>       }
>       char buffer[17];
>       sprintf(buffer, "%ud %02uh %02um %02us", dys, hrs, mnts, dur);
>       return {buffer};
>     icon: mdi:clock-start
>     update_interval: 5s
> 
> switch:
>   - platform: restart
>     icon: mdi:reload-alert
>     name: "Restart Device"
> ```

KinDR007, 2025-08-01:
github://KinDR007/VictronMPPT-ESPHOME@main has included both mptts and bmv or smart shunts ... you dont need other external components . only github://KinDR007/VictronMPPT-ESPHOME@main .

closed 2025-08-01 by KinDR007